tjboldt/Apple2-IO-RPi #10: Programming the eeprom via an apple 2

resend2, 2021-04-26:
Exploring TJBoldt's Applesoft writer for programming the eeprom to apple2-io-rpi v4.  Any objection if I attach/share your applesoft code and my noob-to-project notes here, Terence?

resend2, 2021-04-26:
Ah, found equivalent code in Apple2-IO-RPi.hdv, so I'm hoping you wouldn't mind.

Are lines 30 and 40 (peek and if a<>b) verifying the write and 10ms delay are done?

tjboldt, 2021-04-26:
Yes, the source for writing the firmware is on the drive image. It has a bug though that it doesn't set the current firmware page so it's hit and miss as to which page it writes to. It also doesn't include the entire firmware, only the first page. I will update it one night after work this week. The firmware for the AT28C64 has four pages of firmware for each slot. Even after fixing the bug, the code to update the firmware will only work for the slot that it is in.

tjboldt, 2021-05-09:
See https://github.com/tjboldt/Apple2-IO-RPi/issues/7 

The drive image has been updated to include the Applesoft BASIC program UPDATE.FIRMWARE along with the firmware binary. If you have a working system, you can run this directly from the Apple2-IO-RPi to update the firmware. If not, use other Apple II tools to copy the files to your Apple II and run it there.

closed 2021-05-09 by tjboldt