tjboldt/Apple2-IO-RPi #15: Apple ii/ii+ Uppercase issue

Moddkid, 2021-06-16:
cannot execute linux commands with uppercase only computers.  need workaround for this?

tjboldt, 2021-06-16:
Thanks for bringing this to my attention. The board was developed with an enhanced //e. The current firmware makes some assumptions that lowercase is supported and even expecting an 80 column card to be in slot 3. Although it would be hard to see with uppercase only character ROM, it might be possible to alternate between uppercase and lowercase using the ESC key in the input. This will require a custom input routine instead of the standard ROM based one currently used. This could be a challenge for firmware space on the card. Another alternative would be to always convert text to lower case if a ][ or ][+ is detected. Most Linux commands are lowercase but this won't always work. More of a temporary workaround. Another possibility is to make the command execution a program on the drive image as this allows for enough code space for a custom input routine.

tjboldt, 2021-10-14:
Reopening as only a workaround was implemented and not full shift-key-mod support.