tjboldt/Apple2-IO-RPi #38: Bug:

tjboldt, 2021-11-06:
Edge triggering of GPIO doesn’t work consistently on RPi ZW2 and slow on RPi ZW. Code was changed to use busy wait loops and polling. 

**To Reproduce**
See older code with edge triggering. 

**Expected behavior**
Should be just as fast and consistent edge triggering as it is polling. 

**Hardware (please complete the following information):**
As described above.

timboldt, 2021-11-06:
Some notes and hypotheses:
1) periph.io uses the older sysfs interface, which is deprecated. It might be worth trying the newer gpiod interface. https://embeddedbits.org/new-linux-kernel-gpio-user-space-interface/
2) The A2IO code that sets up the edge detection (e.g. `inWrite.In(gpio.PullDown, gpio.BothEdges)`) isn't checking the return status. Since this is an I/O operation, it might be failing silently on the RPi ZW2. Worth adding some error checking.
3) I haven't tried it yet, but `gpiomon` looks like a cool tool for monitoring kernel GPIO events.

tjboldt, 2021-11-09:
Thanks for the research. I might switch to this library in the future: https://github.com/warthog618/gpio For now, the busy-wait loop is working on the Zero W and Zero W 2.

PR #83 (2022-02-16): Replace periph.io with rpio
Change periph.io to rpio as periph.io is using the deprecated sysfs where rpio uses the newer gpiomem. Although this doesn't reintroduce edge detection yet, it does partially resolve #38 as this is no longer using deprecated code.
closed 2022-02-16 by tjboldt