tjboldt/Apple2-IO-RPi #42: Enhancement: SOIC PCB version

bfranske, 2021-11-16:
**Is your feature request related to a problem?**
Yes, DIP parts (especially 74LS245 and AT28C64B) are out of stock from many suppliers with very long lead times.

**Describe the solution you'd like**
SOIC footprint version of the PCB

tjboldt, 2021-11-20:
Indeed it seems that components are increasingly difficult to source. I'm working on a hybrid board that will take either DIP or SOIC so any chip can be of either package type depending on availability.

pmansell20, 2021-12-14:
Just built 2 of these, no 74LS245 DIP in stock anywhere, so used 74HCT245, no issues on ROM 1 IIGS or Laser 128EX so far.
The T in the IC part no is important, HCT parts have switching thresholds similar to LS (that's what the data sheets say anyway).

FWIW No problems to get the AT28C64B DIP in Australia - element14 show them in stock (although they take a couple of days to come into the local warehouse). But there's not a lot of them.

In any case thanks for the easy storage solution for the II's.

Hubidrei, 2022-09-15:
So i could source  all components but the 74LVC374, only found them on Mouser but shipping is too expensive. So i now tried with 74AC374. They seem to work in my BASIS108, will have to do more long time tests.

tjboldt, 2024-08-21:
Note that there is a SOIC version of the A2Pico card which is now compatible with this project.

https://github.com/rallepalaveev/a2pico/tree/main/A2Pico.v2.7

closed 2024-08-21 by tjboldt