tjboldt/Apple2-IO-RPi #43: Enhancement: Add 5v disable jumper to PCB

bfranske, 2021-11-16:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Add a jumper position (or switch position) to disconnect the 5v supply line to the Pi.

**Additional context**
This would allow use of Pis which require more power than the Apple 2 bus can provide by allowing them to be powered separately.

tjboldt, 2021-11-20:
The original intention of this project was to make a standalone board. From the Apple II's perspective, it should be a simple peripheral card, not a separate computer with power cables running to it. That said, it's not difficult to add jumpers to the design. The previous revision of the board had jumpers for selecting a 27C256 EPROM or AT28C64B EEPROM. Given the current scarcity of Zero W and Zero W 2 variants of the Raspberry Pi, it would be nice to have the flexibility to use other boards with external power. The jumpers can be added along with the new hybrid SOIC/DIP board. Note that full sized Raspberry Pi boards do not physically fit with the Apple II's lid closed so you'll need a 40 pin ribbon cable to move the RPi off the card.

bfranske, 2021-11-21:
Recognizing there are some simplicity of assembly reasons for not requiring a jumper alternatives to a full scale jumper could be a spot for a 0 ohm SMD resistor, for a through hole wire link which could be cut, or a spot for a jumper which has a trace between the pads which could be cut through but later replaced with a jumper or a wire link if the user desired to re-connect.

dlxmax, 2022-01-24:
Along the lines of using a more power-hungry rpi, you can undervolt, underclock, and disable some unnecessary functionality to significantly reduce power draw.
https://www.reddit.com/r/raspberry_pi/comments/aki1is/reducing_power_of_the_raspberry_pi_3_b/

wyatt-wong, 2022-02-12:
> Along the lines of using a more power-hungry rpi, you can undervolt, underclock, and disable some unnecessary functionality to significantly reduce power draw. https://www.reddit.com/r/raspberry_pi/comments/aki1is/reducing_power_of_the_raspberry_pi_3_b/

I think the WiFi and BlueTooth functions is required for the time being, especially WiFi for connecting to the Internet. I am not use if @tjboldt would like to BlueTooth functionality or not in existing or future updates.

tjboldt, 2022-02-12:
Wifi is required to set the clock, get code updates, allow ssh to work etc. but the current setup instructions for setting up from scratch do disable Bluetooth.

tjboldt, 2022-02-16:
Some notes on using external power:

1. The power must not be connected when inserting or removing the card from the slot
2. Since the latch chips controlling firmware pages run on 3.3V power supplied by the RPi, the card will hang on boot if the RPi is not powered
3. In testing a v5 board with a power trace cut, there were occasional boot failures while using external power but could be a coincidence
4. Using a USB power meter, the RPi Zero W 2 was typically drawing 170 mA but occasionally peaked around 250 mA, confirming the theory that when using internal power, the card uses up about half of all available power to all slots which Apple claims is a maximum of 500 mA. External power is likely required for systems with other power hungry cards installed.

tjboldt, 2022-02-18:
Created a new sixth prototype with 5V jumper

closed 2022-02-18 by tjboldt