tjboldt/Apple2-IO-RPi #51: Enhancement: Add Nibble/DSK Image support

bfranske, 2021-12-10:
**Is your feature request related to a problem? Please describe.**
Depends on how you define problem :) The problem is Apple2-IO-RPi does not currently support Nibble/DSK images which are useful for software which can't be easily loaded onto a ProDOS volume.

**Describe the solution you'd like**
Implement Nibble/DSK support for DOS 3.3 images, see suggestions on method below.

**Additional context**
I'm a bit naive with Apple II design...but I think this is doable with software/firmware only changes? Did some research and it looks like an approach similar to the open source [Pseudo-Disk II](http://alexfreed.com/FPGApple/pseudodisk.html) would be possible. That project supports both a ProDOS and Nibble mode. The general approach to Nibble support is described as "Boot code identical to one found in a standard DISK II controller card is mapped into the Cx00-CxFF memory space. AVR interprets reads from the addresses responsible for disk operation. For example it notices attempts to position the head and adds a offset to the beginning of the image." In the case of the Apple2-IO-RPi it would seem that the Boot ROM could load the Disk II card code into memory, then a driver on the Pi side (instead of the AVR used by Pseudo-Disk II) could interpret the calls for disk operation and read/write from a Nibble (or DSK) file.

tjboldt, 2021-12-10:
The Raspberry Pi running Linux is not particularly well suited for realtime. The Pseudo-Disk II concept really requires different hardware with a microcontroller. This request would be a good candidate for a Raspberry Pi Pico but I believe someone else is already working on a project for that. This project was designed for command and response, the Apple II sending commands and the Raspberry Pi responding. The focus going forward is to increase what those commands are from the existing ReadBlock, WriteBlock, Execute, Shell, LoadFile etc. and add on things like LoadImage, Compile, WGet, VNC, etc.

tjboldt, 2022-01-14:
Moved to Discussions here: https://github.com/tjboldt/Apple2-IO-RPi/discussions/68

closed 2022-01-14 by tjboldt