tjboldt/Apple2-IO-RPi #59: WGET command directly from ProDOS BASIC

tjboldt, 2022-01-12:
### What do suggest for improvement?

Currently there is already an RPI command that allows Linux commands to be executed on the RPI and output to the Apple II from ProDOS BASIC programs and command prompt but "RPI wget [URL]" will just download to the RPi itself. It would be nice to have files download directly to the Apple II like "WGET [URL] [FILENAME], A$[ADDRESS], T[TYPE]" for example, "WGET https://some-apple2-site.com/files/launcher.system LAUNCHER.SYSTEM, A$2000, TSYS".

timriker, 2022-07-25:
does curl <url> work? it should write to stdout. Is there any way to redirect it's output?

tjboldt, 2022-07-25:
Yes curl works. No, there isn't currently a way to redirect output in the shell app running on the Apple II. You can of course redirect output on the Linux command line but that would stay on the RPi.

tjboldt, 2023-01-08:
Added dynamic drive support so the ProDOS volume is the contents of the current working directory on Linux. This allows for a more direct download. Note that right now the drive only refreshes on change of working directory so you would need to `RPI cd ..` and 'RPI cd [directory where you were]`. After that, 'CAT' will show you the files just downloaded.

closed 2023-01-08 by tjboldt