tjboldt/Apple2-IO-RPi #61: Support for Reactive Micro RAMWorks IIII v1.2 and A2Heaven Ramworks III VGA extender ?

wyatt-wong, 2022-01-12:
Does Apple 2 IO RPi board support Reactive Micro RAMWorks IIII v1.2 (which offers 80 column) and A2Heaven Ramworks III VGA extender ?

Reactive Micro RAMWorks IIII v1.2
https://www.reactivemicro.com/product/ramworks-iiii/

A2Heaven Ramworks III VGA extender
http://www.a2heaven.com/webshop/index.php?rt=product/product&product_id=152

tjboldt, 2022-01-12:
I don’t have either of those to test with but it shouldn’t be a problem as the 80 column firmware is built into the Apple //e ROM. As long as the total current draw on the slots and power supply isn’t too high, it should in theory be fine. My SHELL command is calling $c300 to activate the 80 column mode and uses the standard soft switches for 80 column text page RAM access.

wyatt-wong, 2022-01-12:
When is the board available for sale ?

tjboldt, 2022-01-13:
Closing this issue as it was moved to the Discussions section.

closed 2022-01-13 by tjboldt