tjboldt/Apple2-IO-RPi #72: Unable to boot up Apple 2 IO RPi in Slot 4

wyatt-wong, 2022-02-05:
I found Apple 2 IO RPi was unable to boot up when I installed it in Slot 4, here is my hardware configuration:

Platinum IIe

Slot 7 - A2Heaven Apple II VGA Scaler (https://a2heaven.com/webshop/index.php?rt=product%2Fproduct&product_id=142)
Slot 6 - Apple 5.25 Disk Controller Card with DB19 connector
Slot 5 - Apple II 3.5 Disk Controller Card with DB19 connector (original one, NOT the clone version from Reactive Micro)
Slot 4 - Empty
Slot 3 - VidHD
Slot 2 - Empty
Slot 1 - Apple II Super Serial Card II

No Slot Clock v1.1 from Manila Gear

tjboldt, 2022-02-05:
My first guess would be too much current draw. Does it boot with other cards removed?

wyatt-wong, 2022-02-05:
> My first guess would be too much current draw. Does it boot with other cards removed?

Check my other post. I guessed it was not due to many cards installed, but probably it could not boot up due to I installed it to Slot 4 instead of Slot 5-7. But normally Apple II will auto scan the slots from Slot 7 back to Slot 1, so it can support a maximum of 14 Disk Drives with 7 Disk ][ or Apple 5.25 Disk Controller Cards installed to all the slots.

tjboldt, 2022-02-05:
It should boot in any slot as long as it has enough power. The Apple II bus is limited to 500 mA and the card with RPi draw about 200 mA. It's also possible that you got an EEPROM from one of my development boards which could have older or incorrect firmware. If you can get the AT28C64.BIN file and UPDATE.FIRMWARE onto another drive, you can reinstall the firmware for slot 4. The firmware is programmed per slot.

tjboldt, 2022-02-05:
Note that with related issue of it crashing on boot in other slots, I strongly suspect it's current draw. Try with all other cards removed except 80 column card. If it's still failing, it could be a damaged card.

wyatt-wong, 2022-02-05:
> It should boot in any slot as long as it has enough power. The Apple II bus is limited to 500 mA and the card with RPi draw about 200 mA. It's also possible that you got an EEPROM from one of my [...]

I will try to update the firmware to the EEPROM and see if I can boot it up in slot 4 or not.

tjboldt, 2022-02-05:
If you have an EPROM programmer, you can reprogram the entire chip with contents of AT28C65B.bin. UPDATE.FIRMWARE is an AppleSoft BASIC program for writing the firmware in-system for people who don't own an EPROM programmer. When programming in-system, it only writes the firmware for the slot that the card is in. With an EPROM programmer, you are writing the full firmware for all slots.

wyatt-wong, 2022-02-05:
> If you have an EPROM programmer, you can reprogram the entire chip with contents of AT28C65B.bin. UPDATE.FIRMWARE is an AppleSoft BASIC program for writing the firmware in-system for people who [...]

Since I cannot bootup Apple 2 IO RPi probably, I have only one choice to re-programme the whole EEPROM AT28C64B

tjboldt, 2022-02-05:
I still suspect power issues rather than firmware. Before reprogramming the firmware, try removing all the other cards from the system and see if it works on its own.

wyatt-wong, 2022-02-05:
> I still suspect power issues rather than firmware. Before reprogramming the firmware, try removing all the other cards from the system and see if it works on its own.

Initially when I put it in slot 4, it didn't boot at all. So I have removed the Apple 5.25 Disk Controller from slot 6 and put Apple 2 IO RPi to slot 6 and it failed to boot up successfully.

Removing all the cards for testing is possible but since I have the connectors attached to the backpane of Apple IIe, removing all of them for testing is quite troublesome. I would not try to remove the connectors from the backpane of IIe, but removing all cards without removing the connectors is quite a challenge as I need to ensure the cards will not accidentally fall back into my IIe motherboard and cause a short circuit.

I would like how many cards that other people normally installed in their Apple IIe. If Apple 2 IO RPi couldn't boot up due to too many cards already installed, I think it is an issue that need to tackle.

wyatt-wong, 2022-02-05:
I would try to perform the test the other day and see the result.

tjboldt, 2022-02-05:
Thanks for taking the time to help troubleshoot an in-development prototype. Through the process of elimination, we are determining the issue with your card. It was tested and working before it shipped but it's possible a weak solder joint loosened up during shipping. It's possible that it doesn't work with the particular cards in your system. I'm fairly sure I programmed the EEPROM on that card for all slots but it's possible I mixed up with another EEPROM I was doing development on and in-system programming for slot 7 only. So far about 20 others have successfully used cards that they built themselves or I built for them but that's a pretty small sample rate. If you're not satisfied with the card, you can ship it back for a full refund.

wyatt-wong, 2022-02-06:
After some more testing, I found that it was NOT due to the bad solder joint of the Apple 2 IO RPi Card. I suggest to close this issue. Please refer to the next issue that I report.

tjboldt, 2022-02-06:
Thanks for the update. I will close this issue now.

closed 2022-02-06 by tjboldt