tjboldt/Apple2-IO-RPi #73: Apple 2 IO RPi show monitor prompt after loading ProDOS 2.4.2

wyatt-wong, 2022-02-05:
After I removed the Apple 5.25 Disk Controller Card from Slot 6 and installed Apple 2 IO RPi to slot 6, I booted up Platinum IIe and found monitor prompt showed on screen after loading ProDOS 2.4.2

[image]

[image]

[image]

[image]

On one occasion, the ProDOS 2.4.2 screen was shown forever

[image]

Due the the Apple 2 IO RPi is very close to the Apple II VGA Scaler, I added a paper in between to avoid short circuit.

tjboldt, 2022-02-05:
Aside from current draw from other cards, the only other thing I can think of that might cause that is bad solder joints. It's a hand assembled prototype and shipped across the world. Try it with no other cards in the system and if it's still failing, I can send you a refund or a replacement.

wyatt-wong, 2022-02-06:
This case happened most probably due to the Apple 3.5 Disk Controller Card was installed in Slot 5, even when there are no Apple 3.5 Disk Drive was attached. I would suggest to close to issue. Please refer to Issue 14 for other findings.

tjboldt, 2022-02-06:
Thanks for the update. Closing issue.

closed 2022-02-06 by tjboldt