tjboldt/Apple2-IO-RPi #75: Different cases that Apple 2 IO RPi was unable to boot up

wyatt-wong, 2022-02-06:
### What happened?

Apple 2 IO RPi was unable to boot up in the following cases:

Case 1
When either Apple 5.25 Disk Controller Card or Apple 3.5 Disk Controller Card card was installed in Slot 6 or Slot 5, with Apple 2 IO RPi installed in Slot 4 (and most probably true if Apple 2 IO RPi was installed in Slot 1 or Slot 2, refer to case 2 when Apple 2 IO RPi was installed in Slot 3)

In Case 1, even if no Disk Drive is connected to Apple 5.25 Disk Controller Card and/or Apple 3.5 Disk Controller Card, Apple 2 IO RPi still could not boot up.

Case 2
When Apple 2 IO RPi was installed in Slot 3:

1. With 80 column card installed in auxiliary slot, Apple 2 IO RPi immediately display a BASIC prompt
2. Without 80 column card installed in auxiliary slot, Apple 2 IO RPi can start loading the firmware and then showed the ProDOS 2.4.2 splash screen, but then the screen will hung forever with showing the BASIC prompt.

### Hardware Version

Fifth Prototype (Purchased from Terence)

### Where are you seeing this problem?

Boot

### What computer is this running on?

Apple //e (enhanced / platinum)

tjboldt, 2022-02-06:
It makes sense that slot 3 will not work because that space is required for the 80 column firmware. I will update the documentation to note that it should not be installed in slot 3.

The Apple II, as you know, scans from slot 7 down to search for a card with bootable firmware. So in slot 4, the Apple will try to boot from your other drives first. What happens if you have all the cards in place, a 5.25" drive connected with a boot floppy in? Does the computer successfully boot from floppy? After you boot, you could type `PR#4` at the BASIC prompt to boot the A2-IO-RPi and it will start up. Note, if you booted ProDOS 2.4.2 from floppy disk, the A2-IO-RPi drives will be detected even though they were not booted from. The one issue you might have is when ProDOS goes to check the A2-IO-RPi and it wasn't booted from, it may hang the OS while the firmware on the card is waiting for the RPi to start up.

Thanks again for all your help testing.

tjboldt, 2022-02-06:
README has been updated to reflect the fact that this does not work in Slot 3.

wyatt-wong, 2022-02-06:
> It makes sense that slot 3 will not work because that space is required for the 80 column firmware. I will update the documentation to note that it should not be installed in slot 3.
>
> The Apple II, [...]

I understand the basic boot mechanism. I was not surprised that Apple 2 IO RPi doesn't boot when it was installed in slot 4, with Apple 5.25 Disk Controller Card and Apple 3.5 Disk Controller Card installed in Slot 6 and Slot 5 respectively. However, I was NOT aware that I can manually booted up the Apple 2 IO RPi by typing PR#n where n is the slot number. It should be mentioned in the documentation if it wasn't there.

However I wasn't sure why issue #73 happens when Apple 2 IO RPi was installed in Slot 6 with Apple 3.5 Disk Controller Card in Slot 5. According to the boot mechanism, Apple 2 IO RPi should boot up successfully because it is installed in a higher slot number (slot #6) than the Disk Controller Card installed in a lower slot number (slot #5).

wyatt-wong, 2022-02-06:
I would also like to mention that I tested the Apple 2 IO RPi card a couple of more times. When Apple II Super Serial Card II and VidHD are installed, they did NOT interfere with Apple 2 IO RPi at all.

In short, Apple 2 IO RPi failed to boot up when
1) A Disk Controller was installed in Apple IIe, regardless of whether it is installed in a lower or a higher slot number than Apple 2 IO RPi ;or
2) If Apple 2 IO RPi was installed in Slot 3 of Apple IIe.

tjboldt, 2022-02-06:
That is unexpected if the A2-IO-RPi was in slot 6 and the 3.5" controller was in slot 5. That should boot. I just tested that configuration now and it came up fine on mine. I usually have the A2-IO-RPi in slot 7 and 5.25" controller in slot 6 along with an extended 80 column card in the aux slot but dug out the other card and switched it around for this test.

[image]

[image]

tjboldt, 2022-02-06:
README.md updated with notes about using PR# to boot the card.

tjboldt, 2022-02-08:
raspi-config is confusing to use with a keyboard even when a monitor and keyboard is directly connected to the RPi. Left and right arrows will change focus to the buttons and space presses the button. See here for more information on this tool: https://www.raspberrypi.com/documentation/computers/configuration.html

wyatt-wong, 2022-02-15:
> That is unexpected if the A2-IO-RPi was in slot 6 and the 3.5" controller was in slot 5. That should boot. I just tested that configuration now and it came up fine on mine. I usually have the [...]

Actually the card you installed in Slot 5 is the Liron Card. I am talking about the Apple 3.5 Drive Controller Card (also known as the Apple SuperDrive Controller Card)

[image]
[image]

tjboldt, 2022-02-15:
I wasn't familiar will all the different versions of the 3.5" drive controllers, just that the card I had is a genuine Apple card and says '3.5" Floppy Disk Drive Interface Card' on it. Indeed, looking it up, it is commonly referred to as the Liron card. The Apple2-IO-RPi does work in slot 1,2,4,5,6 or 7 but appears to be incompatible with the particular cards you have in your system.

tjboldt, 2022-02-15:
There is one more extreme measure that could be taken but it would require cutting a trace on the board and using external power for the RPi. It would only help if the issue is power related as originally suspected but not confirmed. If you did want to go that route, I can attach photos of the modification I tested. Note that not only does it require modifying the hardware, but this has had very limited testing so far. I booted only a few times with this modification and you would need to be careful about not having power connected until after the card is seated in the powered off Apple II.

tjboldt, 2022-02-16:
@wyatt-wong Would it be okay to close this issue? The original problem of not working in other slots has been fixed with a combination of code update in RPI.COMMAND to auto-detect the slot as well as documentation changes in the README.md to indicate that slot 3 is not supported plus instructions for booting with PR# when not the first boot device. The remaining problem of incompatibility with other cards in place is likely related to #43 or if you prefer, we can open a new issue just for card incompatibility.

wyatt-wong, 2022-02-16:
> @wyatt-wong Would it be okay to close this issue? The original problem of not working in other slots has been fixed with a combination of code update in RPI.COMMAND to auto-detect the slot as well [...]

I think in your tests the Apple 2 IO RPi have "no" issues starting up in any slot except slot 3. However Apple 2 IO RPi still could not startup properly if I have either Apple 5.25 Disk Controller card or Apple 3.5 Drive Controller card or have both cards installed. Anyway I have no comments to close this issue.

wyatt-wong, 2022-02-17:
> I wasn't familiar will all the different versions of the 3.5" drive controllers, just that the card I had is a genuine Apple card and says '3.5" Floppy Disk Drive Interface Card' on it. Indeed, [...]

Reactive Micro have a clone version of the Apple 3.5 Drive Controller Card, but it is not cheap and I do NOT expect a lot of people own this card.

https://www.reactivemicro.com/product/apple-ii-3-5-drive-controller-card/

I will perform some testing by removing Apple 3.5 Drive Controller Card from Slot 5 and install the Apple 2 IO RPi card to slot 5, with the Apple 5.25 Drive Controlller Card in slot 6 to see if Apple 2 IO RPi can successfully boot up or not. Last time I tried  this and it doesn't work but I can re-confirm it.

And referring to your previous image, you only have Apple 2 IO RPi installed in Slot 6 and Liron Card installed in Slot 5. I did not see you installed the Apple 5.25 Disk Controller Card for testing. I have attached the Apple 5.25 Drive Controller Card for your reference. Note that it looks different from the Apple Disk ][ Card.

[image]
[image]

wyatt-wong, 2022-02-17:
> I think in your tests the Apple 2 IO RPi have "no" issues starting up in any slot except slot 3. However Apple 2 IO RPi still could not startup properly if I have either Apple 5.25 Disk Controller [...]

As you mentioned the RPI.COMMAND issue is fixed. I would like to mention that last time I followed your advice to choose the SSID directly from RPI configuration screen using "sudo raspi-config" command. After updated the latest firmware, I executed -RPI.COMMAND and then list the available SSID using the "RPI a2wifi" list command. I found it can only list ONE of my SSIDs on screen.

I have two SSIDs called LinksysXXXX and LinksysXXXX_5GHz where XXXX is a random digits sequences. The last time when I executed "RPI a2wifi", it can only show the "LinksysXXXX" SSID. The "LinksysXXXX_5GHz" SSID cannot be listed. But of course I can re-confirm this with another testing.

tjboldt, 2022-02-17:
The Raspberry Pi Zero W only supports 2.4 GHz wifi so that is expected that it will not be able to see 5 GHz networks.

tjboldt, 2022-02-17:
I have tested with the Apple 5.25” controller in slot 6 and the Apple2-IO-RPi card in slot 1, 2, 4, 5 and 7 successfully.

wyatt-wong, 2022-02-17:
> @wyatt-wong We can keep this issue open if you want. The initial issues that would happen to any machine (RPI.COMMAND from a slot other than 7, undocumented slot 3 incompatibility and undocumented PR# boot) have been fixed. The remaining issue is still most likely power related which was why I was suggesting moving to #43. Would you be willing to try the following tests?
> 
> 1. Remove all cards except the Apple2-IO-RPi card and the drive controller cards. This will help determine if it is an incompatibility with your two controller cards or simply too much total current draw on the system.
> 2. If the above works, modify the Apple2-IO-RPi card by cutting the 5V power trace going to the RPi. Reinstall all your cards. Then plug a micro USB cable into the RPi and finally power up the Apple. See attached photos.
> 
> [image]
> 
> [image]
> 
> If you can confirm this works, I'll send you a new sixth revision prototype card with jumper to select internal or external power.

1. Remove all cards except the Apple2-IO-RPi card and the drive controller cards. This will help determine if it is an incompatibility with your two controller cards or simply too much total current draw on the system.

@tjboldt I could do the test as follows:

Machine: Platinum IIe
PSU: Reactive Micro Universal PSU
Slot 7 - A2Heaven VGA Scaler
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Empty
Slot 3- Empty
Slot 2 - Empty
Slot 1 - Empty
Aux Slot - 80 Column Card

2. If the above works, modify the Apple2-IO-RPi card by cutting the 5V power trace going to the RPi. Reinstall all your cards. Then plug a micro USB cable into the RPi and finally power up the Apple. See attached photos.

No idea how to perform step 2.

tjboldt, 2022-02-17:
To determine the difference between overall power issues versus card incompatibility, we need to test with just those cards. 

I believe you said that the following fails?

Slot 7 - A2Heaven VGA Scaler
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- Empty
Slot 2 - Empty
Slot 1 - Empty
Aux Slot - 80 Column Card

And this succeeds?

Slot 7 - A2Heaven VGA Scaler
Slot 6 - Empty
Slot 5 - Empty
Slot 4 - Apple2-IO-RPi
Slot 3- Empty
Slot 2 - Empty
Slot 1 - Empty
Aux Slot - 80 Column Card

What I would like you to try is:

Slot 7 - Empty
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- Empty
Slot 2 - Empty
Slot 1 - Empty
Aux Slot - 80 Column Card

and…

Slot 7 - Empty
Slot 6 - Empty
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- Empty
Slot 2 - Empty
Slot 1 - Empty
Aux Slot - 80 Column Card

and…

Slot 7 - Empty
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Empty
Slot 4 - Apple2-IO-RPi
Slot 3- Empty
Slot 2 - Empty
Slot 1 - Empty
Aux Slot - 80 Column Card

This last option is what I use for my testing (except I moved the A2-IO-RPi card into 1,2,4,5 and 7 at different times). 

As for modifying the card, if you’re not comfortable with that and are patient, I can get you a sixth revision card with jumper in about a month (takes a while to get the boards made, shipped, assembled and shipped back across the world again). 

My hypothesis here is that the combination of cards is drawing too much power. By isolating this to just the cards necessary to run the Apple2-IO-RPi plus one, we can determine if there is a specific card incompatibility. I will send the updated board with power jumper free of charge as a thanks for your efforts if you perform these tests.

wyatt-wong, 2022-02-23:
> To determine the difference between overall power issues versus card incompatibility, we need to test with just those cards.
>
> I believe you said that the following fails?
>
> Slot 7 - A2Heaven VGA [...]

You are right that I have performed the following testings:

================================
Test Case 1:

Slot 7 - A2Heaven VGA Scaler
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- **VidHD Display Card**
Slot 2 - Empty
Slot 1 - **Apple II Super Serial Card II**
Aux Slot - 80 Column Card

Result: Fail to boot up Apple 2 IO RPi
=================================

================================
Test Case 2:

Slot 7 - A2Heaven VGA Scaler
Slot 6 - Empty
Slot 5 - Empty
Slot 4 - Apple2-IO-RPi
Slot 3- **VidHD Display Card**
Slot 2 - Empty
Slot 1 - **Apple II Super Serial Card II**
Aux Slot - 80 Column Card

Result: Successful to boot up Apple 2 IO RPi
=================================

What you expect me to perform the following test cases 3, 4 and 5. Unfortunately I will **NOT** be able to do these tests since there was no LCD monitor attached to Apple IIe.

================================
Test Case 3:

Slot 7 - Empty
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- Empty
Slot 2 - Empty
Slot 1 - Empty
Aux Slot - 80 Column Card

Result: TBC
=================================

================================
Test Case 4:

Slot 7 - Empty
Slot 6 - Empty
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- Empty
Slot 2 - Empty
Slot 1 - Empty
Aux Slot - 80 Column Card

Result: TBC
=================================

================================
Test Case 5:

Slot 7 - Empty
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Empty
Slot 4 - Apple2-IO-RPi
Slot 3- Empty
Slot 2 - Empty
Slot 1 - Empty
Aux Slot - 80 Column Card

Result: TBC
=================================

 I would instead suggest to perform the revised tests as follows:

================================
Test Case 3a:

Slot 7 - Empty
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- **VidHD Display Card**
Slot 2 - Empty
Slot 1 - **Apple II Super Serial Card II**
Aux Slot - **Reactive Micro RAMWorks IIII v1.1 + A2Heaven RAMWorks III VGA extender**

Result: TBC
=================================

================================
Test Case 4a:

Slot 7 - Empty
Slot 6 - Empty
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- **VidHD Display Card**
Slot 2 - Empty
Slot 1 - **Apple II Super Serial Card II**
Aux Slot - **Reactive Micro RAMWorks IIII v1.1 + A2Heaven RAMWorks III VGA extender**

Result: TBC
=================================

================================
Test Case 5a:

Slot 7 - Empty
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Empty
Slot 4 - Apple2-IO-RPi
Slot 3- **VidHD Display Card**
Slot 2 - Empty
Slot 1 - **Apple II Super Serial Card II**
Aux Slot - **Reactive Micro RAMWorks IIII v1.1 + A2Heaven RAMWorks III VGA extender**

Result: TBC
=================================

[image]
[image]

For Slot 3 and Slot 1, I will test the cases without VidHD Display Card and Apple II Super Serial Card II and another cases with both cards installed.

wyatt-wong, 2022-02-26:
After perform numerous testings, here are my findings:

1. Apple 2 IO RPi prototype 5 DID NOT have conflict with the following boards:
- Reactive Micro RAMWorks IIII v1.1 and A2Heaven RAMWorks III VGA extender
- Apple II Super Serial Card II
- VidHD Display Card
- Apple 5.25 Disk Controller Card
- Apple 3.5 Disk Controller Card (i.e. Apple II SuperDrive Controller Card)
- A2Heaven VGA Scaler (from previous testing)

================================
Test Case 3a:

Slot 7 - Empty
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- VidHD Display Card
Slot 2 - Empty
Slot 1 - Apple II Super Serial Card II
Aux Slot - Reactive Micro RAMWorks IIII v1.1 + A2Heaven RAMWorks III VGA extender

Result: Apple 2 IO RPi can successfully boot up (see note 1)
================================

Test Case 4a and 5a are NOT tested (see note 2)

Add Test Cases 6, 7, 8 as follows:

================================
Test Case 6:

Slot 7 - Empty
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- Empty
Slot 2 - Empty
Slot 1 - Apple II Super Serial Card II
Aux Slot - Reactive Micro RAMWorks IIII v1.1 + A2Heaven RAMWorks III VGA extender

Result: Apple 2 IO RPi can successfully boot up (see note 3)
================================

================================
Test Case 7:

Slot 7 - Empty
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- VidHD Display Card
Slot 2 - Empty
Slot 1 - Empty
Aux Slot - Reactive Micro RAMWorks IIII v1.1 + A2Heaven RAMWorks III VGA extender

Result: Apple 2 IO RPi can successfully boot up (see note 4)
================================

================================
Test Case 8:

Slot 7 - Empty
Slot 6 - Apple 5.25 Drive Controller Card
Slot 5 - Apple 3.5 Drive Controller Card
Slot 4 - Apple2-IO-RPi
Slot 3- Empty
Slot 2 - Empty
Slot 1 - Empty
Aux Slot - Reactive Micro RAMWorks IIII v1.1 + A2Heaven RAMWorks III VGA extender

Result: Apple 2 IO RPi can successfully boot up (see note 5)
================================

Note:
1. From test case 3a, Apple 2 IO RPi would occasionally failed to startup and showed a screen with lots of character @ and checkerbox character.
[image]
[image]
[image]

tjboldt, 2022-02-26:
Thanks @wyatt-wong for this extensive set of tests. Total current draw on the Apple //e's power supply varies depending on the combination of cards installed. While in the Apple2-IO-RPi, the RPi itself is known to draw between 170 mA and 250 mA as per tests using external power and a USB power meter. Pardon the fuzzy photo... It was in an awkward spot when I took the shot and didn't realize it was too close and out of focus.

[image]
 
Since the total current available to all cards is 500 mA according to the Apple ][ Reference Manual, there is barely half of current left for other cards. In the design of this card, I made it so there are multiple pages of 256 byte firmware that are selected using the leftover bits on the 74LVC374 that latch communication flags going to the RPi. Because the RPi GPIO operates at 3.3V and the Apple II operates at 5V, that level shifting was necessary. To get 3.3V power to the chip, I used the 3.3V power output from the RPi itself. So if there are power consumption issues to the rather sensitive RPi (or the RPi is not connected or powered in the case of external power configuration), the latch doesn't operate and prevents the firmware from working. You'll either get hangs or weird crashes that you've observed. Sometimes you'll have enough power for the latch to make the firmware work but the RPi itself will crash due to power drops and then you'll see lots of "......" on boot where the Apple II can't communicate with the driver running on the RPi.

Note that the original Apple branded power supplies generally cut out and reset the computer when current draw was too high. Your aftermarket power supply might act differently.

For your system, I'd recommend either choosing a card combination that works for you or using external power. I'm still waiting on blank PCBs to arrive next week but will ship out a new board as soon as I can. I will leave this issue in open state until you can try at least once with external power using the new board. If it works with the previously failing card combinations, I would argue that the power theory is mostly confirmed. Then it's a matter of documentation of when and how to use external power or to recommend using fewer cards in the system.

tjboldt, 2022-04-13:
@wyatt-wong Did you get a chance to test with external power with the card I sent you?

1. Ensure the power is off
2. Take the RPi from the v5 board and install it on the v6 board
3. Remove the power jumper from the v6 board
4. Install the card into your Apple //e along with the other cards that previously conflicted
5. Connect a USB power supply to the micro USB power input on the RPi
6. Power on your Apple //e

Let me know how it goes.

Thanks,
Terence

wyatt-wong, 2022-04-13:
> @wyatt-wong Did you get a chance to test with external power with the card I sent you?
>
>
>
> 1. Ensure the power is off
>
> 2. Take the RPi from the v5 board and install it on the v6 board
>
> 3. Remove the [...]

No. I haven't tested it yet. But there are 4 consecutive Easter holidays from 15-Apr-2022 and I should have time to do the test.

wyatt-wong, 2022-04-15:
> @wyatt-wong Did you get a chance to test with external power with the card I sent you?
>
> 1. Ensure the power is off
> 2. Take the RPi from the v5 board and install it on the v6 board
> 3. Remove the [...]

> Thanks, Terence

Just a small comment. You may consider to move the power jumper closer near to the SN74LS245N for easier removal

[image]

wyatt-wong, 2022-04-15:
> @wyatt-wong Did you get a chance to test with external power with the card I sent you?
>
> 1. Ensure the power is off
> 2. Take the RPi from the v5 board and install it on the v6 board
> 3. Remove the [...]

I was unable to startup Apple 2 IO RPi after using external power. I tried to use power from Apple IIe to startup Apple 2 IO RPi and the result failed. I detached the Raspberry Pi Zero 2 W and re-attach it to the Apple 2 IO RPi card, first I tried to use the power from Apple IIe and I was unable to see the text of Apple 2 IO RPi and then stopped without loading. When I used the external USB power, I got similar monitor screen.

I will take out the Raspberry Pi Zero 2 W and test it to make sure it is functional

[image]

tjboldt, 2022-04-16:
When the RPi is not connected or not powered, it is expected that the card will not start up correctly as it needs the 3.3V power from the RPi to control the 74LVC374 chip selecting the firmware address.

In the case where the jumper is off the card and the RPi is powered externally by USB, it should start up. If that still fails in your system, it's possible you have a bad AT28C64B as per #102 . I only found that after that the system becomes less stable after updating the firmware multiple times. If you have a copy the SHELL app on another disk, you can try running that and see if it is able to communicate with the card.

wyatt-wong, 2022-04-16:
> When the RPi is not connected or not powered, it is expected that the card will not start up correctly as it needs the 3.3V power from the RPi to control the 74LVC374 chip selecting the firmware [...]

Have you test the card before you ship to me ? I could not startup Apple 2 IO RPi using the power from Apple II (with jumpers on).

tjboldt, 2022-04-16:
Yes, the card was tested in multiple slots on my Apple //e prior to shipping.

wyatt-wong, 2022-04-16:
> Yes, the card was tested in multiple slots on my Apple //e prior to shipping.

In that case, it should not have a BAD AT28C64B

tjboldt, 2022-04-16:
It should be fine but I only discovered later that reprogramming the chip multiple times can cause it to be flaky. It doesn't fail always but becomes less reliable. It's possible it passed fine on mine but made the chip a bit more sensitive. You can swap the chip from the other card to test if you're up for that. That said, your system is quite unique with an aftermarket power supply and numerous modern cards. It's possible that the Apple2-IO-RPi in its current design simply won't work with your system. If you mail the cards back, I will refund you and I'll close this case.

wyatt-wong, 2022-04-16:
> It should be fine but I only discovered later that reprogramming the chip multiple times can cause it to be flaky. It doesn't fail always but becomes less reliable. It's possible it passed fine on [...]

I will try to re-program it using WinBond W27C512 or W27E512 and see the result.

tjboldt, 2023-01-08:
Closing due to inactivity and that this appears to be a one-off case with specific hardware.

closed 2023-01-08 by tjboldt