tjboldt/Apple2-IO-RPi #94: Arrow keys don't work in vi

tjboldt, 2022-02-27:
### What happened?

The arrow keys don't work in vi even though they work in other editors. 

Note, this was a common problem years ago with different terminal editors so there was always a workaround to use hjkl keys to navigate. This workaround still applies here. More info on vi usage can be found [here](https://docs.oracle.com/cd/E19253-01/806-7612/editorvi-43/index.html)

Another workaround is to use an alternative editor.
`sudo apt install nano`

### Hardware Version

Fifth Prototype (Self-assembled)

### Firmware Version

000F

### Driver Version

0023 (replaced periph.io with rpio)

### Where are you seeing this problem?

SHELL command

### What computer is this running on?

Apple //e (enhanced / platinum)

hanshuebner, 2022-02-27:
I think instead of implementing a VT100 emulation in the go server, it'd be better to implement a proprietary terminal emulation (maybe based on the apple2e-p terminfo entry and then use tmux or screen for VT100 emulation if required.

Much of the emulation is already handled by the output routine in the Apple ROM, so it'd require only a few commands to be implemented on the server side.  Putting all of the terminal emulation on the Apple side would also not be terribly hard.

```
chrodon 1014_% infocmp -L apple2e-p
#	Reconstructed via infocmp from file: /usr/share/terminfo/61/apple2e-p
apple2e-p|Apple //e via Pascal,
	auto_left_margin, move_standout_mode,
	columns#80, lines#24,
	bell=^G, clear_screen=\014$<100/>, clr_eol=\035$<4/>,
	clr_eos=\013$<4*/>,
	cursor_address=\036%p2%' '%+%c%p1%' '%+%c,
	cursor_down=^J, cursor_home=^Y, cursor_left=^H,
	cursor_up=^_, enter_reverse_mode=^O,
	enter_standout_mode=^O, exit_attribute_mode=^N,
	exit_standout_mode=^N, init_2string=^R^N,
	key_backspace=^H, key_down=^J, key_left=^H, key_right=^U,
	key_up=^K, newline=\r$<100/>, reset_1string=^R^N,
	scroll_forward=^W, scroll_reverse=^V, tab=^I,
```

I'm planning to go this route with my Lisp server, but maybe you like the idea enough :)

tjboldt, 2022-02-27:
The shell app 6502 code already makes use of a lot of the 80 column firmware, with the Go code translating VT100 codes to Apple control codes. It seems a lot of Linux apps send non-compliant codes and even send colour codes not supported by VT100 even when term=vt100 is set. Parsing escape codes is certainly possible in 6502 assembler but it’s much easier to handle in Golang.

hanshuebner, 2022-02-27:
My point is that there is no need to emulate a VT100 at all, because both screen and tmux already do that.  What it takes to use most Linux command-line tools (i.e. vi, emacs, nano, top, trafshow and all the bunch) is to implement a terminal that is described by a terminfo entry, and the apple2e-p terminfo entry is one such example.  My suggestion was that instead of trying to emulate a VT100 faithfully, adding the few escape codes that the apple2e-p terminfo entry defines would not be hard, and remove the requirement to implement everything that the VT100 terminfo entry defines (look at `infocmp -L vt100` to see what this would be).

hanshuebner, 2022-02-27:
BTW:  The `vttest` program is a pretty good tool that helps to see how good a VT100 emulation is.  It does not use terminfo, but really exercises all features of a VT100.  Not all features can even mapped to terminfo capabilities, so full vttest compatibility is probably not something to thrive for.

It is certainly also possible to use a customized terminfo entry for a terminal emulation that only implements some VT100 features.  Terminfo can also be used to define custom screen sizes, but most issues that revolve around the last column in an emulator not behaving as the Linux programs expect them to do have to do with cursor and position behavior at the end of the line.  Again, terminfo provides for capabilities that tell the program whether the cursor wraps and when, so if it is difficult to implement some behavior in a certain environment, a custom terminfo entry might be a clean solution.

Just things to consider.  Sorry if I'm writing things that you already know.

tjboldt, 2022-02-28:
I did try out vttest during development a few months ago. It tests many things that will never work like double sized fonts and 132 columns. The main focus was to have a generally useable shell. So vi/vim work (albeit with hjkl for navigation), nano makes a working substitute for vi, top works, ssh works, even vitetris works. The terminal info is set (to 79x24 to avoid autoscrolling as noted in another issue). I had considered using screen but the method chosen of converting VT100 to Apple 80 column codes (plus a few custom features) was easier to implement.

tjboldt, 2022-03-01:
@hanshuebner I did some looking into it... It appears that not all of the functionality is covered by changing the terminal type to apple2e-p. Apps like vi set the window to a smaller size for scrolling part of the screen. I handle that now by parsing the VT100 commands, make that a special single byte command (plus parameter bytes), read that in my 6502 shell code and set the Apple II window size accordingly. Also, some apps do not honour the terminal type. Even setting terminal type to vt100, they ignore it and send other non-compliant codes. If you do get it working with terminal type apple2e-p, let me know.

hanshuebner, 2022-03-01:
As far as I have understood from comparing the golang and assembly code and the apple2e-p terminfo entry, some of the emulation is handled by the COUT rom code, and some (like absolute cursor positioning and scrolling region handling) is handled by a combination of the golang code (to parse the ANSI escape sequence) and the assembly code (to translate from the simplified commands to actual cursor movement).  The apple2e-p terminfo entry is probably meant to be used for a particular terminal emulator written in (UCSD?) Pascal which does the same thing - Pass most commands to COUT and handle some in Pascal.  Scrolling region support is not defined in the terminfo entry, but it could be added (the vt100 entry has it as `change_scroll_region=\E[%i%p1%d;%p2%dr`).  The apple2e-p entry should only be seen as a starting point, really.

What programs did not honor the `TERM` setting and instead send vt100 codes?  Those are probably new programs which just assume that the terminal is somewhat ANSI compatible, which is a pretty fair assumption to make in most cases nowadays.  However, with tmux or screen, it should be possible to work with these programs even if the terminal emulation is not an ANSI/vt100 derivative.

I went a bit on a tangent with some other retro stuff, but will return to working on the terminal emulation part in the next days and will report what my results are.

tjboldt, 2022-03-01:
Yes, part of it is Golang and part of it is assembler. In Golang, I set the high bit of the output for the Apple II on all regular text and I convert VT100 codes to the equivalent 80 column firmware control codes. When needed, I send special single byte commands of my own (that are high bit clear), followed by a single byte value parameter. For example, setting the top of the window sends "T" and the binary row number. "B" sets the bottom of the window. "H" sets the horizontal tab. "V" sets the vertical tab etc.

It was months ago so I don't recall which programs sent non-VT100 compliant codes when TERM=vt100 but there were a few. In the code, I see I added a workaround for the "top" command since it tries to vtab to row 25 even when ROWS=24.