tjboldt/Apple2-IO-RPi #95: Push v6 to PCBWay Projects

dlxmax, 2022-03-01:
### What do suggest for improvement?

Please update the files on PCBWay Projects for the sixth prototype.
https://www.pcbway.com/project/shareproject/Apple2_IO_RPi_v5.html

I have the chips on order, so it's time to order PCBs. Would like the latest. Thanks!

maded2, 2022-03-01:
hi,  any chance for you to have both SMD & DIP footprint on the same board, as sourcing the chips is challenge at the moment.

thanks,
eddie

tjboldt, 2022-03-01:
https://www.pcbway.com/project/shareproject/Apple2_IO_RPi_v6_bfa4be82.html

tjboldt, 2022-03-01:
> hi, any chance for you to have both SMD & DIP footprint on the same board, as sourcing the chips is challenge at the moment.
> 
> thanks, eddie

Hi Eddie,

Sorry, no, I haven't had time to design a SOIC version yet. The DIP chips are available still but I had to get them from three different suppliers. A SOIC version will likely get designed later this year. This is covered by #42 and now that the sixth revision is up on PCBWay, I will close this issue.

Regards,
Terence

dlxmax, 2022-03-01:
Thanks, Terence!
However, there might be an issue with the fingers on the connector. They're just PCB trace line size as shown on v6, but correct on v5. Is it a rendering issue at PCBWay, or something else?

tjboldt, 2022-03-01:
I checked the gerber files and they are fine but PCBWay is indeed having issues rendering them. This could be from updating to KiCAD 6 from 5.

commit bd88d194 (2022-03-01): Fix #95 gerber render for PCBWay compatibility
closed 2022-03-01 by tjboldt