tjboldt/Apple2-IO-RPi #105: Update for Raspberry Pi Imager v1.7.2

DaManDOH, 2022-06-18:
## Setup starting from scratch
1. Have PCBs made from [PCBWay](https://www.pcbway.com/project/shareproject/Apple2_IO_RPi_v6_bfa4be82.html) or from the gerber and drill files in the Hardware folder or email me for a blank or fully assembled board
2. Solder chips, header and capacitors in place
3. Attach Raspberry Pi Zero W facing outward from the card
4. Install [Raspberry Pi OS](https://www.raspberrypi.org/software/) on microSD card using a modern computer
    1. Use Raspberry Pi Imager
    2. Click "Choose OS"
    3. Select "Raspberry Pi OS (Other)" => "Raspberry Pi OS Lite (32 bit)"
    4. Click on the gear icon in the bottom right
        1. Enable ssh
        2. Set password
        3. Configure wifi
        4. Set locale settings  
5. Put microSD card in the RPi
6. Install the expansion card into the Apple II
7. Power on the Apple II
8. Install firmware with utility (can be found on Apple2-IO-RPi.hdv drive image) or use EPROM programmer with AT28C64B.bin
9. Use `ssh` to connect to the RPi using the password you configured
10. `wget --no-cache -O - https://raw.githubusercontent.com/tjboldt/Apple2-IO-RPi/main/RaspberryPi/setup.sh | bash`

## Options
You can support two drives and change their drive images.
1. Modify the `ExecStart` line in `/etc/systemd/system/apple2driver.service` and make it something like the following: `ExecStart=/home/pi/Apple2-IO-RPi/RaspberryPi/apple2driver/apple2driver -d1 /home/pi/Apple2-IO-RPi/RaspberryPi/Apple2-IO-RPi.hdv -d2 /home/pi/Apple2-IO-RPi/RaspberryPi/TotalReplay401.hdv`
2. `sudo systemctl daemon-reload`
3. `sudo systemctl restart apple2driver.service`

_Originally posted by @tjboldt in https://github.com/tjboldt/Apple2-IO-RPi/discussions/63_

DaManDOH, 2022-06-18:
See menu change:
[image]

Also includes nested list improvements.

tjboldt, 2023-11-01:
Thanks. I incorporated some of your changes and made some additional changes to support 64 bit on RPi Zero 2 W.

closed 2023-11-01 by tjboldt