tjboldt/Apple2-IO-RPi #108: Need clarification

mgederberg, 2022-09-04:
### What do suggest for improvement?

Before I really attempt this one... Of course the jumper would need to be broken and the pi would need a ribbon cable running outside of the Apple IIe with the pi on it's own juice, but can the card handle a pi4? If so, will the GUI run so I can run stuff like emulationstation?

tjboldt, 2022-09-04:
I have never tested with an RPi 4 but in theory it should work using USB power with the jumper removed. The intention of this project is to provide functionality to the Apple II using the RPi as a source of storage, computation and network. If you are looking for an emulator using the Apple II for its devices, you should look at the Apple2Pi project.

mgederberg, 2022-09-05:
Right on TJ... Thanks for the heads up on that. Does your card work with Schmenk's software though?

tjboldt, 2022-09-05:
No, unfortunately, the two cards are incompatible.

mgederberg, 2022-09-05:
Yeah and that would make sense based on my understanding that Schmenk's software is dependent on the same serial communication as ADTPro. Your design is independent of that and in a sense a more direct connection if I might say. With that said, it would be a neat enhancement to get remote control of an RPI4 with GUI using your card, I think it might even be faster and more reliable than SSC intermediaries, no?