tjboldt/Apple2-IO-RPi #109: Boot problem

amazingcobrazil, 2022-09-04:
### What happened?

Even after preparing the card and writing the AT28C64B several times, the card crashes at the message:
Waiting for RPi FW:000F..._
Sometimes it logs into ProDOS, but most of the time I have to log into the OS and give a command: sudo reboot

### Hardware Version

Sixth Prototype (Self-assembled)

### Firmware Version

000F

### Driver Version

0024 (fix VT100 arrow keys)

### Shell Version

000D (added version info)

### Where are you seeing this problem?

ProDOS access

### What computer is this running on?

Apple //e (enhanced / platinum)

tjboldt, 2022-10-09:
Can you see from the logs on the RPi? Is the apple2driver service crashing? Also, be careful with reflashing the AT28C64B as they have a limited number of write cycles. I have worn out a couple of chips during development.

gwortmann, 2023-06-19:
I had the same problem after I set up the RPI Zero WH with the provided script. I solved the issue by starting the installation from scratch with a stable internet connection.

Checked my installation as follows - on Raspbian:
> sudo systemctl --> showed that the apple2driver failed to start
> sudo journalctl -u apple2driver --> told me, that the folder home/pi/apple2-IO-RPi was completely missing

make sure while setting up the image, that you name the user "pi"