tjboldt/Apple2-IO-RPi #130: Apple basic hangs when accessing the drive after exiting SHELL

benjamink, 2023-01-30:
### What happened?

After entering the RPi shell via `-SHELL` then exiting the RPi shell any further access to the virtual drives causes the Apple to hang.  A Control+Reset will reboot the Apple & the drive will be accessible again.  Interacting with the virtual drive normally doesn't appear to have any issues.  Only after entering the RPi shell & then exiting are there problems.

### Hardware Version

Fifth Prototype (Purchased from Terence)

### Firmware Version

000F

### Driver Version

0028 (add live drive loading and regeneration)

### Shell Version

000D (added version info)

### Where are you seeing this problem?

Other

### What computer is this running on?

Apple //e (enhanced / platinum)

benjamink, 2023-01-30:
Confirmed that loading `-RPI.COMMAND` & running `rpi uname -a` to access the RPi *DOES NOT* cause the Apple to hang on subsequent access.

benjamink, 2023-01-30:
`apple2drive` logs from exiting RPi shell & running `catalog` from Apple BASIC: 

```Jan 29 21:50:54 a2iorpi apple2driver[1539]: stdout closed
Jan 29 21:50:54 a2iorpi apple2driver[1539]: Reset request
Jan 29 21:51:05 a2iorpi apple2driver[1539]: Sending date/time...3D 2E 33 15
Jan 29 21:51:05 a2iorpi apple2driver[1539]: Read block 0002 in slot 7, drive 1...succeeded
Jan 29 21:51:05 a2iorpi apple2driver[1539]: Sending date/time...3D 2E 33 15
Jan 29 21:51:05 a2iorpi apple2driver[1539]: Read block 0002 in slot 7, drive 1...succeeded
Jan 29 21:51:05 a2iorpi apple2driver[1539]: Read block 0006 in slot 7, drive 1...succeeded
Jan 29 21:51:05 a2iorpi apple2driver[1539]: Read block 0007 in slot 7, drive 1...succeeded```

benjamink, 2023-01-30:
Restarting `apple2driver` from the RPi & hitting Control+Reset will drop Apple back to BASIC & the RPi virtual drive will be accessible again.  No Apple reset/reboot required.

benjamink, 2023-01-30:
Found [gellweiler/errgotrace](https://github.com/gellweiler/errgotrace) & tried it on the current code.  I don't understand if this is useful but these are the lines in the log output when exiting the RPi shell: 

```
Jan 29 22:20:22 a2iorpi apple2driver[4634]: stdout closed
Jan 29 22:20:22 a2iorpi apple2driver[4634]: Reset request
Jan 29 22:20:23 a2iorpi apple2driver[4634]: 2023/01/29 22:20:23 [ERRGOTRACE] a2io.A2Gpio.WriteByte: timed out writing byte -- read stuck high
Jan 29 22:20:23 a2iorpi apple2driver[4634]: 2023/01/29 22:20:23 [ERRGOTRACE] a2io.A2Gpio.ReadByte: timed out reading byte -- write stuck high
Jan 29 22:20:23 a2iorpi apple2driver[4634]: 2023/01/29 22:20:23 [ERRGOTRACE] a2io.readCharacter: timed out reading byte -- write stuck high
Jan 29 22:20:23 a2iorpi apple2driver[4634]: 2023/01/29 22:20:23 [ERRGOTRACE] a2io.A2Gpio.ReadCharacter: timed out reading byte -- write stuck high
Jan 29 22:20:24 a2iorpi apple2driver[4634]: 2023/01/29 22:20:24 [ERRGOTRACE] a2io.A2Gpio.ReadByte: timed out reading byte -- write stuck high
Jan 29 22:20:24 a2iorpi apple2driver[4634]: 2023/01/29 22:20:24 [ERRGOTRACE] a2io.A2Gpio.ReadByte: timed out reading byte -- write stuck high
Jan 29 22:20:24 a2iorpi apple2driver[4634]: 2023/01/29 22:20:24 [ERRGOTRACE] a2io.readCharacter: timed out reading byte -- write stuck high
Jan 29 22:20:24 a2iorpi apple2driver[4634]: 2023/01/29 22:20:24 [ERRGOTRACE] a2io.A2Gpio.ReadCharacter: timed out reading byte -- write stuck high
```

tjboldt, 2023-01-30:
The errors that are not normally logged are intentionally not logged. It's a busy wait loop so the timeout waiting for a byte is normal. That said, it looks like the SHELL app sends a reset while the RPi is still in the process of shutting down its side. A less than optimal way to prevent this could be having the SHELL app wait a second before sending the reset command after receiving a null terminator from the RPi. This will have the side effect of a delay on exit. Ideally the RPi would block reads until it is finished shutting down the shell code.

closed 2023-01-31 by tjboldt

benjamink, 2023-01-31:
Awesome!  Confirmed fixed!