tjboldt/Apple2-IO-RPi #141: apple2driver draws 100% cpu

JPT77, 2023-04-02:
### What happened?

Hi, 
apple2driver draws 100% CPU which makes the single core zero extremely slow. and heats the zero and some 74xx.

this is when Apple2-IO-Rpi is not connected to the zero:
```
Read block 0101 in slot 0, drive 1...failed can't write byte while byte is incoming
Read block 0101 in slot 0, drive 1...failed timed out writing byte -- read stuck low
... repeated over and over
```

and this is when working fine, but still drawing 100% cpu

```
Read block 0002 in slot 5, drive 1...succeeded
Sending date/time...7F 2E 26 0D
Read block 0002 in slot 5, drive 1...succeeded
Sending date/time...7F 2E 26 0D
Read block 0002 in slot 5, drive 1...succeeded
Sending date/time...7F 2E 26 0D
Read block 0002 in slot 5, drive 1...succeeded
Read block 0003 in slot 5, drive 1...succeeded
Read block 0004 in slot 5, drive 1...succeeded
Read block 0005 in slot 5, drive 1...succeeded

```
could this be caused by using HCT instead of LS? Wasn't able to get LS for some of the chips.

from Apple point of view it works fine, but SSH and Wifi totally suck on single core Zero.

driver version is 29, not listed..
firmware version F
shell version - no idea which shell and how to find out, sorry.

thank you very much. 

ps. tried other Apple II+ and Apple IIe, looks like the same problem.

### Hardware Version

Sixth Prototype (Self-assembled)

### Firmware Version

000F

### Driver Version

0028 (add live drive loading and regeneration)

### Shell Version

000D (added version info)

### Where are you seeing this problem?

Other

### What computer is this running on?

Apple II+

tjboldt, 2023-04-03:
The high CPU usage is due to polling the GPIO ports. There is a new version of the hardware in the works that will change this.

tjboldt, 2023-04-04:
I should also say, this has nothing to do with the 74LS chips... It's how polling the GPIO works for now. See #127

closed 2023-11-01 by tjboldt

JPT77, 2023-11-01:
Great!
Tested on Zero 2 and Zero 1.
Zero 2 25% of one CPU core.
Zero 1 50% CPU. 
The biggest problem was that SSH just didn't work well with 100% busy on a single core CPU. So I think this is gonna work fine. 
Not sure which Raspberry I used last time, because I cannot connect to my Zero 1. Maybe it's one without wifi. So cannot confirm that it really works better now. Sorry :(

ps. you have to iterate through the update process https://github.com/tjboldt/Apple2-IO-RPi/discussions/65 before you'll see any changes.

tjboldt, 2023-11-01:
Unfortunately I noticed it made keyboard very sluggish and unusable in the shell... So I made another tweak that will go in shortly. It will consume more CPU but not as bad as before the first fix.

commit 8d5aa26b (2023-11-01): Add backoff to reduce CPU usage (fixes #141) (#151)

* Add backoff to reduce CPU usage

* Update driver version for CPU fix