tjboldt/Apple2-IO-RPi #143: Apple2.IO.RPI.hdv - (two) issues

OrangeBox72, 2023-05-22:
### What happened?

On the HDV image (I downloaded on 05/22/2023),

Issue one,
The Basic program "UPDATE.FIRMWARE" seems to be corrupt.
* It shows up in PRODOS w/date (04-MAR-22),
* When I try running it from BitsyBoot, it ABENDs
* When I escape out to BASIC.SYSTEM and load the program,  nothing is there.
* When I use COPYIIPLUS.8.4 (on the HDV), it shows the program as EMPTY (when viewing textually)
* When I use COPYIIPLUS.8.4 (on the HDV), it shows '00 00 00 00 00 00 00 00'  (when viewing values)

Issue two,
In trying to burn the firmware, I hand typed the Basic Program from the gitrepo..
"Apple2-IO-RPi/Apple2/Update.Firmware.bas"

LINE 100 seems to be incorrect..
100  PRINT  CHR$ (4)"BLOAD AT28C64B,A$2000"

The IMAGE FILE on the HDV is  AT28C64B.BIN,  but you reference "AT28C64B"  (sans ".BIN")
attempting to BLOAD as written gives a "PATH NOT FOUND".

I have additional problems, but they are most likely on my side. ;-)

### Hardware Version

Fifth Prototype (Self-assembled)

### Firmware Version

000F

### Driver Version

0028 (add live drive loading and regeneration)

### Shell Version

000D (added version info)

### Where are you seeing this problem?

Other

### What computer is this running on?

Apple //e (enhanced / platinum)

tjboldt, 2023-07-06:
The files are meant to be accessed via the drive itself which admittedly is a little hard if you're just creating one. The dynamic ProDOS volume created by the Apple2-IO-RPi converts the raw ASCII into an AppleSoft BASIC program and the .BIN extension is removed automatically from the AT28C84B.BIN file. I should update the documentation to better explain this.

tjboldt, 2024-05-30:
Documentation updated and firmware added to the release. Closing this issue but if there is something else to change, please reopen this issue or create a new one.

closed 2024-05-30 by tjboldt