tjboldt/Apple2-IO-RPi #145: The 3 logic chips can be replaced by a single PLD

rallepalaveev, 2023-08-30:
### What do suggest for improvement?

A potential improvement and decreasing of the chip count would be to use a PLD instead of the 3 logic chips 7400, 7432, 7432.
I have created a design with ATF22V10 which, I regularly use, should work :). This is based on the schematics posted on 20 Feb 2022 - Rev 0.6.2.

Here is the PLD source code:

Name     A2-IO-RPi;
PartNo   00;
Date     30/08/2023;
Revision 01;
Designer R.Palaveev;
Company  None;
Assembly None;
Location None;
Device   g22v10;

/** Inputs **/

PIN 1 = A0;
PIN 2 = A1;
PIN 3 = A2;
PIN 4 = A3;
PIN 5 = IOSelect;
PIN 6 = DevSel;
PIN 7 = IOStrobe;
PIN 8 = RW;

/** Outputs **/

PIN 14 = WR;
PIN 15 = CE_U0;
PIN 16 = CE_U4;
PIN 17 = Cp_U5;
PIN 18 = CE_U6;
PIN 19 = Cp_U7;

/** Logic Equations **/

WR = !RW; /** to pin 1 U0, pin 22 U1 **/
CE_U0 = IOSelect & DevSel; /** to pin 19 U0 **/
CE_U4 = WR # A0 # DevSel; /** to pin 19 U4 **/
Cp_U5 = RW # A1 # DevSel; /** to pin 11 U5 **/
CE_U6 = WR # A2 # DevSel; /** to pin 19 U6 **/
Cp_U7 = RW # A3 # DevSel; /** to pin 11 U7 **/

tjboldt, 2023-08-30:
What software and hardware do you recommend for programming these chips?

rallepalaveev, 2023-08-30:
I use WinCUPL to write the source code and then compile it to a .jed file. Then I use a TL866-II Plus to burn the chips.
WinCUPL is available on the Microchip website: https://www.microchip.com/en-us/products/fpgas-and-plds/spld-cplds/pld-design-resources#

closed 2023-09-12 by rallepalaveev