tjboldt/Apple2-IO-RPi #147: add read-only jumper please

JPT77, 2023-10-04:
### What do suggest for improvement?

There are circumstances where the first byte of the ROM is overriden. 
I got a report, where this happens when CP/M is booted. 
I myself experience it every now and then.

A simple solution would be to have a jumper that allows or forbids writing to the EEPROM from the apple. 

thank you very much

tjboldt, 2023-10-04:
This is a great suggestion. I'll make up a new board design soon.

JPT77, 2023-10-05:
Until then, here is a workaround that regularly fixes the first byte automatically: 
https://forum.classic-computing.de/forum/index.php?thread/29134-rgbtohdmi-platine-apple2-io-rpi-usw-status/&postID=435751#post435751

closed 2024-06-25 by tjboldt