tjboldt/Apple2-IO-RPi #148: Is the Apple2 IO RPi hardware in gerneral able to replace the ROM?

JPT77, 2023-10-23:
### What do suggest for improvement?

If yes, could you please point me to where I can implement providing the ROM?

Looks like I got more bad ROMs in my Apple ][plus. 

thanks

Hubidrei, 2023-10-23:
As far as i know the RP2040 can act as a rom card.

tjboldt, 2023-10-24:
If you’re asking if this card will replace the computer’s main ROM, the answer is no. It only provides 256 bytes of ROM in the peripheral card space. That said, building a ROM replacement card isn’t too difficult. It’s just not the focus of this project or compatible with this hardware.

JPT77, 2023-10-24:
My idea was to make it in software...

tjboldt, 2023-10-24:
While you can pass a lot of code through the port from the card to the Apple II, it will not occupy the expected ROM space. Definitely needs different hardware to replace the system ROMs. If you mean do calculations on the Pi side and just call the card to run that and get results, that of course will work.

JPT77, 2023-10-25:
Ok. Thanks. That's what I wanted to know :)

closed 2023-10-25 by JPT77