tjboldt/Apple2-IO-RPi #149: Create your own disk on the fly from directory

JPT77, 2023-10-29:
### What do suggest for improvement?

Hi, 

First I wanted to start a discussion, but after analyzing the topic I decided it should be a minor improvement.

I understand that directly booting floppy images is impossible because emulating a floppy drive is impossible out of hardware reasons.

The apple2driver has a mode where it assembles a ProDOS image on the fly from `~/Apple2-IO-RPi/RaspberryPi/driveimage/`

Two issues with this. First it tells: 

    This drive is dynamically generated
    from the current working directory on
    the Raspberry Pi and is in RAM only.

which is not true. it is created from `~/Apple2-IO-RPi/RaspberryPi/driveimage/`

Second: I tried passing a directory as parameter to apple2driver, but this did not work. Enhancement?

Not directly related: 
how do I provide software into this directory so that I can start it? Extract files from floppy images?

thanks

tjboldt, 2023-10-30:
The ProDOS volume is dynamically generated in RAM on the RPi from the source files held in that directory. The ProDOS volume itself if not stored on disk, just the source files in a Linux directory. You can add files to the directory and it will pick them up next time the ProDOS volume is generated or you can change your configuration to use another directory.

tjboldt, 2023-11-21:
Use the RPI.COMMAND to control the drive images, both from image files and dynamic from a Linux directory:

```
]-rpi.command
]rpi a2drive 1 regen ~/Apple2-IO-RPi/RaspberryPi/driveimage
]rpi a2drive 2 load /home/pi/Games.hdv
```

closed 2023-11-21 by tjboldt