tjboldt/Apple2-IO-RPi #153: Shell does not fully work

JPT77, 2023-11-19:
### What happened?

Hi, 

I downgraded my Zero2W to ZeroW and somehow it does not connect to wifi any more. 
So this is the first time I really use the shell. And it's great on the Apple IIe. 

But raspi-config does not work. You can chose an entry but you cannot open it. "Enter" just does nothing. Pressing S or F does not do anything either. 
And I think there are a few more things which do not work. 

Could you please have a look into it?

thanks

ps. regarding the versions you ask for: Shell shows version E - cannot select below.
how do I run RPI a2version? I don't know the driver version. But I updated everything just now following your guide.

### Hardware Version

Sixth Prototype (Self-assembled)

### Firmware Version

000F

### Driver Version

Older (error calling RPI a2version, pre-2022-Feb-07)

### Shell Version

000D (added version info)

### Where are you seeing this problem?

SHELL command

### What computer is this running on?

Apple //e (enhanced / platinum)

tjboldt, 2023-11-19:
I updated the bug template. Missed adding shell 000E a year ago. 

Are you running the latest Raspberry Pi OS? I will see if I can reproduce your issue this weekend.

JPT77, 2023-11-19:
No. Just did an update, but there wasn't much change. Should I upgrade from Debian 11 to Debian 12? Or should I just write a new Raspberry OS image?

tjboldt, 2023-11-19:
I have had problems with just pulling the micro SD card out of a Zero W and putting it into a Zero 2 W but it wasn’t the same issue you’re having. The navigation in the raspi-config was also confusing but that is unrelated and the same in other versions. It has been a while since I tried. Not at my computer now but will have a look when I am.

JPT77, 2023-11-19:
Tried with Raspi 3, same. 
Upgraded to Debian 12, same. 
Realized there are more problems: 
- Enter does not work in NANO menu. have to use VI. 
- editing command lines that are wrapped does not work.

tjboldt, 2023-11-20:
Tested... `raspi-config` with VT100 emulation is a bit hard to navigate but it does work... Up and down arrows to choose what you want, then `Tab` to go to the `<Select>` button on the screen, then press `Space`

Alternatively you can configure wifi without the shell...

`]-rpi.command`
`]rpi a2wifi list`
`]rpi a2wifi select SSID PASSWORD REGION`

JPT77, 2023-11-20:
Ah, it's SPACE! But still, nano does not work. Well. Not really important ;)
It wasn't the wifi. Wifi is up and running, but still no network traffic on Zero1W. Still investigating.
Ah and now I understood the rpi commands. thanks for that ;)

closed 2023-12-05 by tjboldt