tjboldt/Apple2-IO-RPi #172: Garbage from RPI command after update

polileco, 2024-06-07:
### What happened?

Hi,

Thanks for the nice work the project is amazing.

After updating to the firmware 0011 and the 002D driver after booting into ProDOS RPI command does not work and just puts garbage on the screen.

Regards

### Hardware Version

Fifth Prototype (classic hardware without jumper)

### Firmware Version

0010 (classic hardware)

### Driver Version

002D (add support for RPi Pico hardware)

### Shell Version

000D (added version info)

### Where are you seeing this problem?

RPI Command

### What computer is this running on?

Other

tjboldt, 2024-06-07:
Have you rebuilt the RPI.COMMAND and SHELL since upgrading?

`cd ~/Apple2-IO-RPi/Apple2`
`./assemble.sh`

The RPI.COMMAND is likely failing to detect the card as the firmware ID bytes changed.

tjboldt, 2024-06-07:
Oh, and after you reassemble the binaries, you need to refresh the dynamic drive image... Normally I'd suggest using the RPI command to regenerate the drive image but since that's not working for you, maybe on the Linux side, restart the service.

`sudo systemctl restart apple2driver`

Then reboot the Apple II.

polileco, 2024-06-07:
> Have you rebuilt the RPI.COMMAND and SHELL since upgrading?
> 
> `cd ~/Apple2-IO-RPi/Apple2` `./assemble.sh`
> 
> The RPI.COMMAND is likely failing to detect the card as the firmware ID bytes changed.
>
> Oh, and after you reassemble the binaries, you need to refresh the dynamic drive image... Normally I'd suggest using the RPI command to regenerate the drive image but since that's not working for you, maybe on the Linux side, restart the service.
> 
> `sudo systemctl restart apple2driver`
> 
> Then reboot the Apple II.

Yes. Just for a sanity check did it again, same result.

Running 'rpi ls' from prompt, for example, I got this:

[image]

Let me know if I can assist you or if more info is needed

Tks

polileco, 2024-06-07:
Oh, 'rpi.command' is executed fine and find the card. The problem arise with the execution of 'rpi' command.

[image]

Regards

tjboldt, 2024-06-07:
I haven't been able to reproduce the issue the way you're seeing it. I did however find a bug that if you `-rpi.command` after it's already loaded, it echos back in an infinite loop when you try and use `rpi ls` after. I assume it's only loaded once (by the startup.bas)?

polileco, 2024-06-07:
> I haven't been able to reproduce the issue the way you're seeing it. I did however find a bug that if you `-rpi.command` after it's already loaded, it echos back in an infinite loop when you try and use `rpi ls` after. I assume it's only loaded once (by the startup.bas)?

Yes. Letting my tests to bare bone. Only the IO-RPi on slot 6 and using de default drive image from the project.

tjboldt, 2024-06-07:
Okay, put everything exactly the way you have it in slot 6 with no other cards other an extended 80 column card and I see the problem. I'll try and make a fix as soon as possible.

polileco, 2024-06-07:
> Okay, put everything exactly the way you have it in slot 6 with no other cards other an extended 80 column card and I see the problem. I'll try and make a fix as soon as possible.

Tks! Let me know if i can help in some way.

With IO-RPi in another slot the problem persist? I can try this weekend.

Regards

tjboldt, 2024-06-07:
Side note... It was working in slot 7 as that is the default... The code that was setting the slot number on detection was off by 256 bytes because the detection code got bigger and pushing into the next page. I had to fix the offset by one.

PR #173 (2024-06-07): Fix #172 garbage from RPI.COMMAND
Fixes #172
commit 9786590f (2024-06-07): Fix #172 garbage from RPI.COMMAND (#173)

* Fix #172 garbage from RPI.COMMAND

* Update version number
closed 2024-06-07 by tjboldt