tjboldt/Apple2-IO-RPi #181: Meta key in -SHELL

sethkush, 2024-08-08:
### What do suggest for improvement?

Would it be possible to make one of the apple keys work as meta when using -SHELL?
Currently Esc works, but it's not as good as a dedicated meta key.

tjboldt, 2024-08-09:
Thanks for the explanation... Currently the project is using VT100 terminal emulation rather than something like xterm-256color etc. I don't see how I'd send the alt/meta key escape code in VT100. Maybe I'm missing something. It's possible to support other terminal emulation but it would be a lot of work. Here are the VT100 codes I'm aware of:

https://espterm.github.io/docs/VT100%20escape%20codes.html

If you can find how to escape the ALT key codes in VT100, adding code to detect the Apple key is easy. Otherwise, I can start looking at other terminal emulation in the future.

sethkush, 2024-08-24:
According to this https://xn--rpa.cc/irl/term.html the alt key actually just sends ESC quickly followed by whatever it's combined with, so I think it would be simple to implement within the VT100 escape codes since ESC already works.

tjboldt, 2024-08-28:
@sethkush Thanks for the explanation and links. I tried sending ESC-character from the Apple II assembly code but this apparently wasn't fast enough for the terminal to pick it up as a meta character. Instead, when the Open-Apple key is pressed, I send the character without stripping the high bit. On the RaspberryPi end, if it sees the high bit set, it clears it and then sends ESC plus the character. This seems to be working for me now.

PR #185 (2024-08-28): Add support for meta key
Closes #181
Use the Open-Apple key to send meta key (alt)
commit 2837dd34 (2024-08-28): Add support for meta key

Closes #181
Use the Open-Apple key to send meta key (alt)
closed 2024-08-28 by tjboldt