tjboldt/Apple2-IO-RPi #187: issue on boot

g91, 2024-10-11:
### What happened?

[image]
hello my friend I'm curious if you've ever seen this before and how to fix it this is happening whenever the machine is turned on and the card is attempted to be accessed. machine was working fine for many hours turned it off came back turned it back on then this issue. c702- a=20 x=00 y=FF P=B5 S=F9

### Hardware Version

Seventh Prototype (lastest classic hardware)

### Firmware Version

0011 / 8011 (add clock driver)

### Driver Version

0030 (add support for meta key)

### Shell Version

0011 (add support for meta key)

### Where are you seeing this problem?

Boot

### What computer is this running on?

Apple //e (enhanced / platinum)

tjboldt, 2024-10-12:
Depending on which other cards and software you run, it is possible to accidentally overwrite the firmware on the Apple2-IO-RPi card. Since you have a v7 board, you can change the jumpers to put the EEPROM in read-only mode. I'd recommend trying to rewrite the firmware, then set the jumpers for read-only mode. Actually, you might want to try that with only the A2IORPi card in the system to ensure it doesn't get messed up. Then after it's read-only, put everything back in and fire it up.

g91, 2024-10-13:
I bought the card pre-assembled can you tell me of a good eprom programmer that I could use.

tjboldt, 2024-10-13:
If the card has the AT28C64B chip on it, you can actually use the Apple II to program the firmware, although it will only program for the slot that the card is located in. After rewriting the firmware, I'd recommend setting the jumper to make it read-only.

g91, 2024-10-14:
would it be possible to get the Flasher onto a floppy image   I did something wrong and now I'm getting this error after shell      Apple2-IO-RPi not found

tjboldt, 2024-10-14:
It looks like it's still booting and you have a valid drive image. The `Apple2-IO-RPi` not found error is likely caused by corrupt firmware bytes. Try running `-UPDATE.FIRMWARE`, then set the jumper to read-only mode after. Also, best to try this with no other cards in the system in case there is a conflict and something else tries to write to the EEPROM space.