tjboldt/Apple2-IO-RPi #188: RPi FW stuck on OK

omigeot, 2025-01-10:
### What happened?

Upon booting on the RPi card's slot, it waits for RPi FW (0011), then prints OK then ... that's all, it's stuck on this for ages. If I ssh to the Pi, the logs only show heaps of : `apple2driver[303]: Read block 0000 in slot 6, drive 1...succeeded` (always on block 0000). The apple2driver process is stuck at 100% CPU.

A little context : I've got my IIGS back from the basement and it hasn't run in a while. So I updated Apple2-IO-RPi while I was at it. Didn't work, ran setup.sh again on the Pi. Booted but kept stuck on "Apple2-IO-Rpi not found" ; according from a related issue's details, I ran -UPDATE.FIRMWARE again, and this time everything worked flawlessly for a couple hours.

Then, out of nowhere, after one restart, I got the current behaviour, described up there.

Note : since I can't boot on Rpi anymore, I'm not sure how I could check a2version or the shell version...

### Hardware Version

Sixth Prototype (classic hardware power jumper)

### Firmware Version

0011 / 8011 (add clock driver)

### Driver Version

0030 (add support for meta key)

### Shell Version

0011 (add support for meta key)

### Where are you seeing this problem?

Boot

### What computer is this running on?

Apple IIgs

tjboldt, 2025-01-12:
Block 0000 is the ProDOS boot loader. I can imagine one of two things failing there. Either something overwrote part of your firmware or if you’re using a disk image rather than the in-memory dynamic generated disk, maybe something overwrote that block on your image. There have been a few people with issues of firmware being damaged so in the latest revision of the hardware, a jumper was added to set the EEPROM to read only mode.

omigeot, 2025-01-12:
So, if I'm understanding right : block 0000 could either (A) come from a file on the RPi's filesystem, or (B) from the EEPROM on the board, right?

If that's so, I suppose I could easily test whether it's (A) by just changing the file it's pointing to (in my case, it's a symlink, so an extra-easy task). If that's not improving situation, it's probably (B), and my EEPROM content is damaged. Which leads me to following question : what are my ways to fix it? is it doable from the board, or do I need extra-hardware?

tjboldt, 2025-01-13:
Block 0000 is not on the EEPROM, it's either in a file on your RPi or generated in the RPi's RAM depending on how you configured it. Let me know if you need help figuring that out. The EEPROM might be fine but it's possible it's corrupted and preventing it from loading the full block and executing it. If the EEPROM is at fault, you can reprogram it without special hardware but you'll need to be able to get the firmware and firmware updater onto a floppy disk to run it there. Unfortunately that's easier to do if you already had the A2IORPi running. There are other creative ways to get files over to an Apple II but will take some effort.

omigeot, 2025-01-13:
> Unfortunately that's easier to do if you already had the A2IORPi running.

Am I wrong, or would I be able to re-flash the EEPROM using something like that? https://github.com/gfoot/picoprom

(which, as far as I'm concerned, is basically "without special hardware")

felangga, 2025-02-22:
Same here, mine stuck on OK then do nothing. 
The RPI log only shows this logs
```
Feb 22 03:18:20 raspberrypi apple2driver[330]: Generating Drive in memory from: /home/felangga/Apple2-IO-RPi/RaspberryPi/driveimage
Feb 22 03:18:21 raspberrypi apple2driver[330]: Starting Apple II RPi v0030...
Feb 22 03:18:21 raspberrypi apple2driver[330]: Starting communications via GPIO ports...
```